devbisme/skidl #172: [SKiDL BUG] ref_prefix Part parameter does not get applied if the ref_repfix parameter is already defined in a lib

Timot05, 2022-12-12:
**Describe the bug**
The ref_prefix Part parameter does not get applied if the ref_repfix parameter is already defined in a lib

**To Reproduce**
1. Define a part: `c = Part(lib="Device.kicad_sym", name="C_Small", ref_prefix="test", dest=TEMPLATE)`
2. Instantiate the part: `c_test = c()`
3. generate the setlist: `generate_netlist()`

-> The part name is `C1`

**Expected behavior**
I'm hoping the part name would be `test1` instead as setting the parameter would overwrite the default defined in the lib.

**Desktop:**
 - MacOS
 - Python version: python 3.10
 - SKiDL version: 1.1.0

**Additional context**
Adding this piece of code after copying the parameters from the library could fix it. Happy to implement if that makes sense.
```
if ref_prefix:
            self.ref_prefix = ref_prefix
```

devbisme, 2022-12-14:
Thanks for finding this error! I incorporated the fix in the `development` branch. Try it out and let me know if it works OK for you.

Timot05, 2022-12-24:
Yep, that worked. Thank you!

closed 2022-12-24 by Timot05