devbisme/skidl #179: [SKiDL BUG]incompatable with latest  KiCad v7

Zhuoli, 2023-02-28:
**Describe the bug**
KiCad released new version "KiCad 7.0.0" on Feb/12/2023,  skidl's part method call on KiCad-7.0.0 would result in KeyError: "F0" 

**To Reproduce**
Steps to reproduce the behavior:
1. Install KiCad version 7, news [here](https://www.kicad.org/blog/2023/02/Version-7.0.0-Released/)
2. Set environment variable "$KICAD_SYMBOL_DIR" to "/Applications/KiCad/KiCad.app/Contents/SharedSupport/symbols/"
3. Run example code on skidl's [readme](https://github.com/devbisme/skidl)
4. See error
`Traceback (most recent call last):
  File "skidl101.py", line 7, in <module>
    r1, r2 = 2 * Part("Device", 'R', TEMPLATE, footprint='Resistor_SMD.pretty:R_0805_2012Metric')
  File "/Users/zhuoli/.pyenv/versions/3.6.15/lib/python3.6/site-packages/skidl/part.py", line 194, in __init__
    part = lib[name].copy(dest=TEMPLATE)
  File "/Users/zhuoli/.pyenv/versions/3.6.15/lib/python3.6/site-packages/skidl/schlib.py", line 192, in get_part_by_name
    parts.parse(get_name_only)
  File "/Users/zhuoli/.pyenv/versions/3.6.15/lib/python3.6/site-packages/skidl/part.py", line 373, in parse
    parse_func(get_name_only)
  File "/Users/zhuoli/.pyenv/versions/3.6.15/lib/python3.6/site-packages/skidl/tools/kicad/kicad.py", line 297, in parse_lib_part
    _parse_lib_part_kicad_v6(self, get_name_only)
  File "/Users/zhuoli/.pyenv/versions/3.6.15/lib/python3.6/site-packages/skidl/tools/kicad/kicad.py", line 817, in _parse_lib_part_kicad_v6
    self.ref_prefix = self.fields["F0"]  # Part ref prefix (e.g., 'R').
KeyError: 'F0'`

**Expected behavior**
No error should raised from skidl's official example.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS : Mac os
 - Python version: 3.10.4
 - SKiDL version: unknown
 

**Additional context**
The failure is brought by this [merge](https://github.com/devbisme/skidl/commit/50e3bf6f69114fb626818b699fd16bf6633b5637#diff-1cc25423444869702efa94b0f6f33f7b3f05a5c5330451a5860bbcc09f94fff4) with a hardcode logic to search part in kicad **v6**

Zhuoli, 2023-02-28:
A workaround to this F0 error is to manually download v6 Schematic symbols [here](https://gitlab.com/kicad/libraries/kicad-symbols/-/tree/6.0.11), and set $KICAD_SYMBOL_DIR to location of this v6 Schematic symbols

usman510, 2023-06-05:
[image]

Proposed FIX: Change "F0" to "Reference"

I believe the field lookup for KiCAD v7 libraries has changed the field name for the reference designator to "Reference"

jakeogh, 2023-06-25:
> Proposed FIX: Change "F0" to "Reference"

That worked here. skidl latest with kicad-7.0.2

dlalagroup-thabo, 2023-08-14:
I am trying to check #182, most things seems to be handled there, just not sure.

Even when changing "F0" to "Reference" will allow parts to load, there is more error/s in '__getattr__'.

when debugging `self.__getattribute__("fields")` is empty.

```
@export_to_all
class SkidlBaseObject(object):

    # These are fallback lists so every object will have them to reference.
    erc_list = list()
    erc_assertion_list = list()

    def __init__(self):
        self.fields = {}

    def __getattr__(self, key):
        try:
            # Don't use super()!! It leads to long run times on Python 2.7.
            return self.__getattribute__("fields")[key]
        except KeyError:
            raise AttributeError
```
[image]

devbisme, 2023-08-17:
Which version of SKiDL are you using? Is it 1.2.1 from PyPi? I would suggest trying the development version:
```bash
pip install git+https://github.com/devbisme/skidl@development
```
The development version includes some new code for handling KiCad 6 and 7. It's not quite ready for release. For one thing, it needs documentation. You have to use `KICAD6` or `KICAD7` as follows:
```python
set_default_tool(KICAD7)
```
Send me the code causing the problem and I'll try it out.

xiaophil, 2023-08-22:
Hello, I have installed the development version of SKiDL as described above but the KICAD7 name is not defined.
I cannot see this symbol anywhere in the development branch of the repo.
What am I missing?

devbisme, 2023-08-24:
My apologies! When I finally got around to checking this, I found I'd failed to push the development branch to Github! I've done that, so give it a try and we can move on to whatever problem comes next...

cyberhuman, 2023-09-10:
I installed the development version, set `KICAD7_SYMBOL_DIR`, called `set_default_tool(KICAD7)` but there is still an issue:
```
Python 3.11.3 | packaged by conda-forge | (main, Apr  6 2023, 08:57:19) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from skidl import *
WARNING: KICAD6_SYMBOL_DIR environment variable is missing, so the default KiCad symbol libraries won't be searched. @ [/home/john/<frozen importlib._bootstrap_external>:940=>/home/john/<frozen importlib._bootstrap>:241]
>>> set_default_tool(KICAD7)
>>> search('ESP32')
Traceback (most recent call last):                                             
  File "<stdin>", line 1, in <module>
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/part_query.py", line 130, in search_parts
    for part in search_parts_iter(terms, tool):
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/part_query.py", line 106, in search_parts_iter
    lib = SchLib(
          ^^^^^^^
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/schlib.py", line 88, in __init__
    tool_modules[tool].load_sch_lib(
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/tools/kicad7/lib.py", line 140, in load_sch_lib
    parent_part = lib[item[1]]
                  ~~~^^^^^^^^^
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/schlib.py", line 209, in __getitem__
    return list_or_scalar(self.get_parts_by_name(id))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/schlib.py", line 203, in get_parts_by_name
    part.parse(partial_parse)
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/part.py", line 388, in parse
    tool_modules[self.tool].parse_lib_part(self, partial_parse)
  File "/home/john/.conda/envs/doe/lib/python3.11/site-packages/skidl/tools/kicad6/lib.py", line 331, in parse_lib_part
    pin_func = pin_io_type_translation[pin[1].value().lower()]
               ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'free'
```

devbisme, 2023-09-10:
This problem occurred because I erroneously referenced KICAD6 in the KICAD7 library parser. I fixed it in the development branch. Give it a try and see if it works for you.

cyberhuman, 2023-09-11:
Hi @devbisme! I tried the updated version and the error is the same, although it is in the kicad7/lib.py now 😅 
```
Python 3.11.3 | packaged by conda-forge | (main, Apr  6 2023, 08:57:19) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from skidl import *
WARNING: KICAD_SYMBOL_DIR environment variable is missing, so the default KiCad symbol libraries won't be searched. @ [/home/raman/<frozen importlib._bootstrap_external>:940=>/home/raman/<frozen importlib._bootstrap>:241]
WARNING: KICAD6_SYMBOL_DIR environment variable is missing, so the default KiCad symbol libraries won't be searched. @ [/home/raman/<frozen importlib._bootstrap_external>:940=>/home/raman/<frozen importlib._bootstrap>:241]
>>> set_default_tool(KICAD7)
>>> search('opamp')
Traceback (most recent call last):                                             
  File "<stdin>", line 1, in <module>
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/part_query.py", line 130, in search_parts
    for part in search_parts_iter(terms, tool):
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/part_query.py", line 106, in search_parts_iter
    lib = SchLib(
          ^^^^^^^
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/schlib.py", line 88, in __init__
    tool_modules[tool].load_sch_lib(
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/tools/kicad7/lib.py", line 140, in load_sch_lib
    parent_part = lib[item[1]]
                  ~~~^^^^^^^^^
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/schlib.py", line 209, in __getitem__
    return list_or_scalar(self.get_parts_by_name(id))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/schlib.py", line 203, in get_parts_by_name
    part.parse(partial_parse)
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/part.py", line 388, in parse
    tool_modules[self.tool].parse_lib_part(self, partial_parse)
  File "/home/raman/.conda/envs/cadquery/lib/python3.11/site-packages/skidl/tools/kicad7/lib.py", line 331, in parse_lib_part
    pin_func = pin_io_type_translation[pin[1].value().lower()]
               ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'free'
```

devbisme, 2023-09-11:
KiCad 7 introduced a new pin type: free. I added that to the development branch. Should clear this error.

cyberhuman, 2023-09-12:
@devbisme that helped, thank you!

Jubei-VA, 2024-01-18:
I'm not sure if i'm running into the exact same issue.
I'm on Ubuntu, Kicad7

```python

import os
os.environ["KICAD7_SYMBOL_DIR"] = "/usr/share/kicad/symbols/"

from skidl import *

set_default_tool(KICAD7)

# Create part templates.
q = Part("Device", "Q_PNP_CBE", dest=TEMPLATE)
r = Part("Device", "R", dest=TEMPLATE)

# Create nets.
gnd, vcc = Net("GND"), Net("VCC")
a, b, a_and_b = Net("A"), Net("B"), Net("A_AND_B")

# Instantiate parts.
gndt = Part("power", "GND")             # Ground terminal.
vcct = Part("power", "VCC")             # Power terminal.
q1, q2 = q(2)                           # Two transistors.
r1, r2, r3, r4, r5 = r(5, value="10K")  # Five 10K resistors.

# Make connections between parts.
a & r1 & q1["B C"] & r4 & q2["B C"] & a_and_b & r5 & gnd
b & r2 & q1["B"]
q1["C"] & r3 & gnd
vcc += q1["E"], q2["E"], vcct
gnd += gndt

generate_netlist()
```
Results in errors:

```

WARNING: Could not load KiCad schematic library "Device", falling back to backup library. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:11]
WARNING: Could not load KiCad schematic library "Device", falling back to backup library. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:12]
WARNING: Could not load KiCad schematic library "power", falling back to backup library. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:19]
WARNING: Could not load KiCad schematic library "power", falling back to backup library. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:20]
ERROR: No footprint for GND/#PWR1. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for VCC/#PWR2. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for Q_PNP_CBE/Q1. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for Q_PNP_CBE/Q2. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for R/R1. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for R/R2. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for R/R3. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for R/R4. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
ERROR: No footprint for R/R5. @ [/home/jubei/Documents/Work-Syncthing/python/skidl-test/skidl_tb.py:31]
INFO: 0 warnings found while generating netlist.
INFO: 9 errors found while generating netlist.
```

I am using the development version. My environment variable for KICAD7_SYMBOL_DIR was never set. However when I open Kicad7, I can go to the symbols and they work fine over there.

```

jakeogh, 2024-03-15:
Tested on KiCad v8, same issue with the missing footprints.

```
# eix-installed all | grep kicad
sci-electronics/kicad-8.0.0
sci-electronics/kicad-footprints-8.0.0
sci-electronics/kicad-meta-8.0.0
sci-electronics/kicad-packages3d-8.0.0
sci-electronics/kicad-symbols-8.0.0
sci-electronics/kicad-templates-8.0.0

# ls
total 3.5K
 512 drwxr-xr-x 2 user user    3 2024-03-15 16:09:00 .
1.5K drwxr-xr-x 3 user user   10 2024-03-15 16:01:01 ..
1.5K -rwxr-xr-x 1 user user 1.2K 2024-03-15 16:08:34 example_from_readme_bug_179.py

# cat example_from_readme_bug_179.py 
#!/usr/bin/env python3

import os
os.environ["KICAD_SYMBOL_DIR"] = "/usr/share/kicad/symbols/"
os.environ["KICAD_3DMODEL_DIR"] = "/usr/share/kicad/3dmodels/"
os.environ["KICAD_FOOTPRINT_DIR"] = "/usr/share/kicad/footprints/"
os.environ["KICAD_TEMPLATE_DIR"] = "/usr/share/kicad/template/"

from skidl import *

set_default_tool(KICAD)
# uncommenting these makes no difference
#lib_search_paths[KICAD].append('/usr/share/kicad/symbols')
#footprint_search_paths[KICAD].append('/usr/share/kicad/footprints')

# Create part templates.
q = Part("Device", "Q_PNP_CBE", dest=TEMPLATE)
r = Part("Device", "R", dest=TEMPLATE)

# Create nets.
gnd, vcc = Net("GND"), Net("VCC")
a, b, a_and_b = Net("A"), Net("B"), Net("A_AND_B")

# Instantiate parts.
gndt = Part("power", "GND")             # Ground terminal.
vcct = Part("power", "VCC")             # Power terminal.
q1, q2 = q(2)                           # Two transistors.
r1, r2, r3, r4, r5 = r(5, value="10K")  # Five 10K resistors.

# Make connections between parts.
a & r1 & q1["B C"] & r4 & q2["B C"] & a_and_b & r5 & gnd
b & r2 & q1["B"]
q1["C"] & r3 & gnd
vcc += q1["E"], q2["E"], vcct
gnd += gndt

generate_netlist()
#generate_pcb()

# ./example_from_readme_bug_179.py 
ERROR: No footprint for GND/#PWR1. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for VCC/#PWR2. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for Q_PNP_CBE/Q1. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for Q_PNP_CBE/Q2. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R1. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R2. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R3. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R4. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R5. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
INFO: 0 warnings found while generating netlist.
INFO: 9 errors found while generating netlist.

ERROR: No footprint for GND/#PWR1. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for VCC/#PWR2. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for Q_PNP_CBE/Q1. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for Q_PNP_CBE/Q2. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R1. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R2. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R3. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R4. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]
ERROR: No footprint for R/R5. @ [/poolz1_1x4TB_A/cfg/python/skidl/bug_179/example_from_readme_bug_179.py:37]

# ls
total 9.0K
1.5K drwxr-xr-x 2 user user    7 2024-03-15 16:09:18 .
1.5K drwxr-xr-x 3 user user   10 2024-03-15 16:01:01 ..
 512 -rw-r--r-- 1 user user    0 2024-03-15 16:09:17 example_from_readme_bug_179.erc
1.0K -rw-r--r-- 1 user user 2.1K 2024-03-15 16:09:18 example_from_readme_bug_179.log
1.5K -rw-r--r-- 1 user user 4.8K 2024-03-15 16:09:18 example_from_readme_bug_179.net
1.5K -rwxr-xr-x 1 user user 1.2K 2024-03-15 16:08:34 example_from_readme_bug_179.py
1.5K -rw-r--r-- 1 user user 2.1K 2024-03-15 16:09:18 example_from_readme_bug_179_lib_sklib.py
```

devbisme, 2025-11-11:
Apologies for my very late reply. I'm going through and clearing some issues.

You have the footprint search paths set, but you still need to specify the particular footprint for each part:
```python
 r = Part("Device", "R", footprint='Resistor_SMD:R_0603_1608Metric', dest=TEMPLATE)
```

closed 2025-11-11 by devbisme