devbisme/skidl #188: Generate schematics for newer versions of KiCad

devbisme, 2023-07-15:
The current version of SKiDL (1.2.0) only generates schematic files for KiCad 5. Please add support for versions 6 & 7.

alphaai, 2023-07-29:
Hi,

Any update on this as I am using kicad 7 and I hope to use skidl as well. Thanks

devbisme, 2023-07-29:
I'm working on it.

aifitrading, 2024-01-08:
waiting for this new feature.

aifitrading, 2024-01-29:
@devbisme any chance to let us know when the new schematic generation will be ready? Even a half baked version will be very very helpful.

jemathews0, 2024-02-01:
I had put some work in on this back in March and made this pr: https://github.com/devbisme/skidl/pull/182 but I was looking for some feedback. I had gotten the `generate_svg` functionality to work but I had questions about generating schematics.

tapegoji, 2024-06-18:
Hey, 
I tried to use your commit, but had issues. Is your code updated? 
I have a code that works for generating svg. Do you think we can merge those and make it a working PR?

devbisme, 2024-06-19:
@jemathews0 PR for generating SVG was submitted while I was making a bunch of changes to SKiDL over an extended period of time, which made it difficult for me to integrate his PR. I've been working on doing that and refactoring the `src/tools` code to support KiCad versions 5, 6, 7, 8, ... I've just released the most current code for that into the `development` branch. I'd be happy to cooperate in getting SVG generation working for any of the 6, 7, or 8 versions of KiCad. (It should be easy to port to the other versions.) I think my port of @jemathews0 stuff is close, but maybe you have something better.

tapegoji, 2024-06-20:
@devbisme I will get working on that.

Rui-Chun, 2025-04-23:
Hi @devbisme , thank you for this great repo and all the efforts.

I am also in desperate need for this new feature. Could you please share any latest updates on this? As mentioned above, a half baked version would already be super helpful. Thanks!

cyberhuman, 2025-09-21:
Hi @devbisme.
Anything here that could use some help?
Recently I needed to provide schematic for an existing design (in KiCAD 9) and it was quite tricky.
However I managed to do it manually by dropping some symbols in eeschema, then editing the file (copy-pasting and updating the UUIDs to match those generated by SKiDL). I also had to create separate files for subcircuits, as UUIDs of parts inside a subcircuit are formatted for hierarchical sheets. It worked but it was quite cumbersome, so it would be nice to have some way to at least get the symbols available on the schematic sheets. Connections can be done later out manually (and don't really matter in some cases as PCB->schematic import adds net names everywhere).

devbisme, 2025-09-21:
Wow, that process you went through sounds like a pain!

I think if we went to the trouble of making a half-assed solution that only places the parts on hierarchical sheets, then actually making a complete solution with the wiring would be pretty easy.

If you want to scope out the effort, take a look at [the code for creating a KiCad5 EESCHEMA file.](https://github.com/devbisme/skidl/blob/development/src/skidl/tools/kicad5/gen_schematic.py) The `node_to_eeschema()` subroutine needs to be re-done for KiCad9. That's approximately 500 lines of code. I wonder how much Github Copilot could help with this and whether the package [kicad-skip](https://github.com/psychogenic/kicad-skip/tree/main) would be useful.

Let me know if you want to try the port to KiCad 9. I can offer advice if not assistance.

cyberhuman, 2025-10-04:
#270