devbisme/skidl #190: [SKiDL BUG] 1.2.0 doesn't work with the previous files

HubertHQH, 2023-07-26:
**Describe the bug**
SKidl 1.2.0 doesn't work with the previous py files. With the old skidl, the py file can run and generate netlist successfully.

    Traceback (most recent call last):
      File "D:\pythonProject1\testgensch.py", line 4, in <module>
        q = Part("Device", "Q_PNP_CBE", dest=TEMPLATE)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\site-packages\skidl\part.py", line 191, in __init__
        lib = SchLib(filename=libname, tool=tool)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\site-packages\skidl\schlib.py", line 88, in __init__
        tool_modules[tool].load_sch_lib(
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\site-packages\skidl\tools\kicad\kicad.py", line 80, in load_sch_lib
        f, _ = find_and_open_file(
               ^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\site-packages\skidl\utilities.py", line 793, in find_and_open_file
        return urllib.request.urlopen(link), link
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 216, in urlopen
        return opener.open(url, data, timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 519, in open
        response = self._open(req, data)
                   ^^^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 541, in _open
        return self._call_chain(self.handle_open, 'unknown',
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 496, in _call_chain
        result = func(*args)
                 ^^^^^^^^^^^
      File "C:\Users\admin\AppData\Local\Programs\Python\Python311\Lib\urllib\request.py", line 1419, in unknown_open
        raise URLError('unknown url type: %s' % type)
    urllib.error.URLError: <urlopen error unknown url type: d>
    
    Process finished with exit code 1

**To Reproduce**
Steps to reproduce the behavior:
1. Install the latest Skidl 1.2.0
2. Set environment variable: "KICAD_SYMBOL_DIR=D:\Program Files\KiCad\6.0\share\kicad\symbols"
3. Run the following code:

```
from skidl import *

# Define components
in_terminal = Net('IN')
out_terminal = Net('OUT')
gnd = Net('GND')
regulator = Part('Regulator_Linear', 'LM1117-3.3', footprint = 'Package_TO_SOT_THT:TO-220-3_Vertical')
cap1 = Part('Device', 'C', value='10uF', footprint = 'Resistor_SMD:R_0805_2012Metric')
cap2 = Part('Device', 'C', value='1uF', footprint = 'Resistor_SMD:R_0805_2012Metric')

# Connect components
in_terminal += regulator['VI']
out_terminal += regulator['VO']
gnd += regulator['GND']
cap1[1] += in_terminal
cap1[2] += gnd
cap2[1] += out_terminal
cap2[2] += gnd

# Set regulator parameters
regulator.value = 'LM1117MP-3.3'

# Generate netlist
generate_netlist()
```
4. See error

**Expected behavior**
Netlist is generated correctly without error.

**Desktop :**
 - Windows 10
 - Python version: 3.11
 - SKiDL version [1.2.0]

**Additional context**
I also tried the example file in readme and get the same issue. But both files work with the previous version Skidl.
Suspect this is environment setting issues or related to KiCad symbol version? Previous Skidl only works with Kicad 6 symbols but not K7. Any idea?

HubertHQH, 2023-07-27:
Hi Dave, the `development` branch addresses my issue. But when I try to `generate_schematic()`, there's another exception.

devbisme, 2023-07-28:
Are you trying to create a schematic? Are you trying to create a schematic while using KiCad 6 or 7 symbols? That won't work. Schematic creation currently only works with KiCad 5 symbol libraries.

HubertHQH, 2023-07-28:
Aha, it's KiCad 5 symbol related. Now everything is fine. Thanks Dave!

closed 2023-07-28 by HubertHQH