devbisme/skidl #202: generate_svg() - issue

sureshmoon, 2024-01-13:
Hello,
Thank you for your wonderful work. I'm currently facing an issue with SKiDL when trying to use generate_svg(). I would appreciate any guidance on how to resolve these warnings and errors.
KICAD-5. 

Code used:
from skidl import *

# Create net stubs.
e, b, c = Net("ENET"), Net("BNET"), Net("CNET")
e.stub, b.stub, c.stub = True, True, True

# Create transistor part template.
qt = Part(lib="Device.lib", name="Q_PNP_CBE", dest=TEMPLATE)

# Instantiate transistor with various orientations.
for q, tx in zip(qt(8), ['', 'H', 'V', 'R', 'L', 'VL', 'HR', 'LV']):
    q['E B C'] += e, b, c  # Attach stubs to transistor pins.
    q.symtx = tx  # Assign orientation to transistor attributes.
    q.ref = 'Q_' + tx  # Place orientation in transistor reference.
generate_svg()

Error:
ERROR: No footprint for Q_PNP_CBE/Q_. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_H. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_V. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_R. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_L. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_VL. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_HR. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
ERROR: No footprint for Q_PNP_CBE/Q_LV. @ [C:\Users\abhic\OneDrive\Documents\SKiDL\main.py:15]
Traceback (most recent call last):
  File "C:\Users\abhic\OneDrive\Documents\SKiDL\main.py", line 15, in <module>
    generate_svg()
  File "C:\Users\abhic\OneDrive\Documents\SKiDL\venv\Lib\site-packages\skidl\circuit.py", line 999, in generate_svg
    subprocess.Popen(
  File "C:\Users\abhic\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\abhic\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1538, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified

roboPanda69, 2024-07-23:
Hey I just faced this issue in my system too. I was able to solve it. It would be majorly a problem for users working on Windows.

In the circuit.py file of skidl module, in the function of generate_svg(self, file_=None, tool=None, layout_options=None), there is a subprocess line as -  subprocess.Popen(["netlistsvg", json_file, "--skin", skin_file, "-o", svg_file], shell=False,)

We have to modify it to subprocess.Popen(["netlistsvg.cmd", json_file, "--skin", skin_file, "-o", svg_file], shell=False,)
i.e., add .cmd to netlistsvg.

From my understanding I am thinking that this lines calls for netlistsvg (you can't directly call netlistsvg because netlistsvg is typically a separate program, not a Python library. It's usually a Node.js application that's installed independently. Python can't directly call JavaScript functions, so it needs to run netlistsvg as an external process.). On Windows, many of the Node.js "binaries" are actually suffixed with the .cmd filename extension. Hence we added that .cmd in our filename.

I refered this  "https://stackoverflow.com/a/50045443/26488957" for my solution.

PS - I am completely new to open source contribution and learning day by day. This is my first comment. If I wrote something wrong then do let me know, I would improve it. 

Thanks

devbisme, 2024-07-24:
Thanks for identifying this problem and showing how to fix it!

I've created a new branch (`version1`) that may fix the problem. Please install it as follows:
```bash
pip install git+https://github.com/devbisme/skidl@version1
```
Let me know if this fixes the problem.

Thanks!

roboPanda69, 2024-07-24:
> Hey I just faced this issue in my system too. I was able to solve it. It would be majorly a problem for users working on Windows.
>
> In the circuit.py file of skidl module, in the function of [...]

Hi sureshmoon, also the transistor "Q_PNP_CBE" is not present in the newer versions of kicad symbols and footprints libraries. Hence we are getting the error for "No footprint for Q_PNP_CBE".

roboPanda69, 2024-07-24:
> Thanks for identifying this problem and showing how to fix it!
> 
> I've created a new branch (`version1`) that may fix the problem. Please install it as follows:
> 
> ```shell
> pip install git+https://github.com/devbisme/skidl@version1
> ```
> 
> Let me know if this fixes the problem.
> 
> Thanks!

Hello devbisme, thanks for the update. I have tested the branch (version1) and it is working correctly. It is generating the svg file now. 

This is the file on which I tested (for reference), and it generated the svg file -

`from skidl import *
set_default_tool(KICAD8)  # Possible values are KICAD5, KICAD6, KICAD7, KICAD8.

# Create input & output voltages and ground reference.
vin, vout, gnd = Net('VI'), Net('VO'), Net('GND')

# Create two resistors.
r1, r2 = 2 * Part("Device", 'R', TEMPLATE, footprint='Resistor_SMD.pretty:R_0805_2012Metric')
r1.value = '1K'   # Set upper resistor value.
r2.value = '500'  # Set lower resistor value.

# Connect the nets and resistors.
vin += r1[1]      # Connect the input to the upper resistor.
gnd += r2[2]      # Connect the lower resistor to ground.
vout += r1[2], r2[1] # Output comes from the connection of the two resistors.

# Or you could do it with a single line of code:
# vin && r1 && vout && r2 && gnd

# Output the netlist to a file.
generate_svg(tool=KICAD8)`

closed 2024-07-25 by devbisme