devbisme/skidl #215: [SKiDL BUG] NotImplementedError when trying to run generate_svg()

doctor-yes, 2024-06-28:
**Describe the bug**
NotImplementedError when trying to run the example code with generate_svg() on https://devbisme.github.io/skidl/. 
netlistsvg is also installed per instruction: npm install https://github.com/nturley/netlistsvg

This is the code: (it keeps complaining about footprint is missing, since the goal here is to just generate the schematic in svg, I removed some component to get things going)

	# Create part templates.
	r = Part("Device", "R", dest=TEMPLATE, footprint='Resistors_SMD:R_0805')

	# Create nets.
	gnd, vcc = Net("GND"), Net("VCC")
	a, b, a_and_b = Net("A"), Net("B"), Net("A_AND_B")

	# Instantiate parts.
	# gndt = Part("power", "GND")  # Ground terminal.
	# vcct = Part("power", "VCC")  # Power terminal.
	# q1, q2 = q(2)  # Two transistors.
	r1, r2, r3, r4, r5 = r(5, value="10K")  # Five 10K resistors.

	# Make connections between parts.
	a & r1 & r4 &  a_and_b & r5 & gnd
	b & r2
	r3 & gnd

Error: 

Traceback (most recent call last):
  File "D:\test.py", line 396, in <module>
    generate_svg()
  File "C:\python\site-packages\skidl\circuit.py", line 993, in generate_svg
    f.write(self.generate_netlistsvg_skin(net_stubs=net_stubs))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\python\site-packages\skidl\circuit.py", line 704, in generate_netlistsvg_skin
    part_svg[name] = part.generate_svg_component(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\python\site-packages\skidl\part.py", line 1054, in generate_svg_component
    return tool_modules[tool].gen_svg_comp(self, symtx=symtx, net_stubs=net_stubs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\python\site-packages\skidl\tools\kicad\kicad.py", line 367, in gen_svg_comp
    return v6.gen_svg_comp(part, symtx, net_stubs=None)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\python\site-packages\skidl\tools\kicad\v6.py", line 319, in gen_svg_comp
    raise NotImplementedError
NotImplementedError

**To Reproduce**
Steps to reproduce the behavior:
As above

**Expected behavior**
Generate svg of the circuit

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Windows 10
 - Python version 3.12.0
 - SKiDL version 1.2.1

**Additional context**
Add any other context about the problem here.

devbisme, 2024-06-29:
Creating SVG schematics isn't supported for KiCad versions 6 and above. This feature is currently being worked on and should be in the development branch in a few weeks.

Thanks for trying SKiDL!

tapegoji, 2024-08-27:
I was able to generate svg successfully. Thank you Dave.