devbisme/skidl #221: [SKiDL BUG] Errors running the example on colab

Kreijstal, 2024-08-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://colab.research.google.com/drive/133ftiEI1jES1ZS2gLGpU78s0XRhFkmPJ?usp=sharing
2. Execute all the steps 
```py
!pip install git+https://github.com/devbisme/skidl.git
!sudo add-apt-repository --yes ppa:kicad/kicad-8.0-releases
!sudo apt update
!sudo apt install kicad

import os
os.environ["KICAD_SYMBOL_DIR"] = "/usr/share/kicad/symbols/"
os.environ["KICAD_3DMODEL_DIR"] = "/usr/share/kicad/3dmodels/"
os.environ["KICAD_FOOTPRINT_DIR"] = "/usr/share/kicad/footprints/"
os.environ["KICAD_TEMPLATE_DIR"] = "/usr/share/kicad/template/"

from skidl import *

set_default_tool(KICAD8)

# Create part templates.
q = Part("Device", "Q_PNP_CBE", dest=TEMPLATE)
r = Part("Device", "R", dest=TEMPLATE)

# Create nets.
gnd, vcc = Net("GND"), Net("VCC")
a, b, a_and_b = Net("A"), Net("B"), Net("A_AND_B")

# Instantiate parts.
gndt = Part("power", "GND")             # Ground terminal.
vcct = Part("power", "VCC")             # Power terminal.
q1, q2 = q(2)                           # Two transistors.
r1, r2, r3, r4, r5 = r(5, value="10K")  # Five 10K resistors.

# Make connections between parts.
a & r1 & q1["B C"] & r4 & q2["B C"] & a_and_b & r5 & gnd
b & r2 & q1["B"]
q1["C"] & r3 & gnd
vcc += q1["E"], q2["E"], vcct
gnd += gndt

generate_netlist(tool=KICAD8) # Create KICAD version 8 netlist.
```
3. See error

```
ERROR: No footprint for GND/#PWR1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for GND/#PWR1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for VCC/#PWR2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for VCC/#PWR2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for Q_PNP_CBE/Q1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for Q_PNP_CBE/Q1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for Q_PNP_CBE/Q2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for Q_PNP_CBE/Q2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R3. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R3. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R4. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R4. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R5. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R5. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
INFO: 0 warnings found while generating netlist.
INFO:skidl:0 warnings found while generating netlist.
INFO: 9 errors found while generating netlist.

INFO:skidl:9 errors found while generating netlist.

ERROR: No footprint for GND/#PWR1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for GND/#PWR1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for VCC/#PWR2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for VCC/#PWR2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for Q_PNP_CBE/Q1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for Q_PNP_CBE/Q1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for Q_PNP_CBE/Q2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for Q_PNP_CBE/Q2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R1. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R2. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R3. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R3. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R4. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R4. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR: No footprint for R/R5. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
ERROR:skidl:No footprint for R/R5. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-1-c996a8f86776>:32]
```

**Expected behavior**
No error

**Screenshots**

**Desktop (please complete the following information):**
 - OS: ubuntu 20
 - Python version: 3.10.12
 - SKiDL version: git master 8e2ff7c8f13d4401f133ab9f2162733f70baa4bc

Kreijstal, 2024-08-04:
Changing to development version does not change the problem

devbisme, 2024-08-04:
Thanks for reporting this.

The physical footprint for each part has to be specified for inclusion in the netlist so the PCBNEW layout program can route connections between pads. So SKiDL reports if they are missing. But even with these errors, the netlist is still generated, it just doesn't provide enough information to make an actual PCB layout.

There are three ways to suppress these error messages:
1. Specify any kind of footprint string for each part such as: `r = Part("Device", "R", dest=TEMPLATE, footprint="null")`
2. Specify the string for the actual footprint of each part (but you'll have to look them up).
3. Specify an empty footprint handler function as shown [here](https://devbisme.github.io/skidl/#handling-empty-footprints).

None of these are in the introductory example because I didn't want to include too much extraneous detail that might obscure what SKiDL does. My feelings about that may change.

Kreijstal, 2024-08-04:
I now seem to get a different type of error.
```
WARNING: Could not load KiCad schematic library "Device", falling back to backup library. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-7-1a06b98d05bf>:12]
WARNING:skidl:Could not load KiCad schematic library "Device", falling back to backup library. @ [/usr/local/lib/python3.10/dist-packages/IPython/core/interactiveshell.py:3553=>/content/<ipython-input-7-1a06b98d05bf>:12]

---------------------------------------------------------------------------

FileNotFoundError                         Traceback (most recent call last)

[<ipython-input-7-1a06b98d05bf>](https://localhost:8080/#) in <cell line: 12>()
     10 
     11 # Create part templates.
---> 12 q = Part("Device", "Q_PNP_CBE", dest=TEMPLATE)
     13 r = Part("Device", "R", dest=TEMPLATE)
     14 

3 frames

[/usr/local/lib/python3.10/dist-packages/skidl/part.py](https://localhost:8080/#) in __init__(self, lib, name, dest, tool, connections, part_defn, circuit, ref_prefix, ref, tag, pin_splitters, **kwargs)
    185                         lib = skidl.load_backup_lib()
    186                         if not lib:
--> 187                             raise e
    188                     else:
    189                         raise e

[/usr/local/lib/python3.10/dist-packages/skidl/part.py](https://localhost:8080/#) in __init__(self, lib, name, dest, tool, connections, part_defn, circuit, ref_prefix, ref, tag, pin_splitters, **kwargs)
    175                 libname = lib
    176                 try:
--> 177                     lib = SchLib(filename=libname, tool=tool)
    178                 except FileNotFoundError as e:
    179                     if skidl.config.query_backup_lib:

[/usr/local/lib/python3.10/dist-packages/skidl/schlib.py](https://localhost:8080/#) in __init__(self, filename, tool, lib_section, **attribs)
     76             if tool in tool_modules.keys():
     77                 # Use the tool name to find the function for loading the library.
---> 78                 tool_modules[tool].load_sch_lib(
     79                     self,
     80                     filename,

[/usr/local/lib/python3.10/dist-packages/skidl/tools/kicad8/lib.py](https://localhost:8080/#) in load_sch_lib(lib, filename, lib_search_paths_, lib_section)
     97             break
     98     if not f:
---> 99         raise FileNotFoundError(
    100             "Unable to open KiCad Schematic Library File {}".format(filename)
    101         )

FileNotFoundError: Unable to open KiCad Schematic Library File Device
```

Not sure what I am doing wrong..
https://colab.research.google.com/drive/133ftiEI1jES1ZS2gLGpU78s0XRhFkmPJ?usp=sharing
anyone can show in colab how is it supposed to be done?

devbisme, 2024-08-05:
It's not finding the `Device.kicad_sym` file. I don't know why since you seemed to be able to find the symbol files previously.

I'm not that familiar with Google colab and where it stores things. I see in your notebook that you are installing KiCad 8.0 and expecting to find the symbols in `/usr/share/kicad/symbols`. Can you check that there actually is a file called `Device.kicad_sym` in that location? Also, after importing `skidl`, check the path strings in `lib_search_paths[KICAD8]` and see if any of them match where the symbols are stored.