devbisme/skidl #222: Fields are not exported to PCB

Psirus, 2024-08-20:
When I add a field to a part as in the documentation (`lm35.fields['new_field'] = 'new value'`), the field will get exported to the netlist with `generate_netlist()`, but not to the PCB generated with `generate_pcb()`.

Is there an option I am missing? Alternatively, do you know of a way to update the netlist with the pcbnew python API if I simply export both PCB and netlist from skidl?

By the way, `module 'skidl' has no attribute '__version__'`, but pip says I have version 1.2.3 installed.

Lastly, thanks for your work, this is such a cool project.

devbisme, 2024-08-23:
It appears there is a shortcoming in the kinet2pcb utility/library: it only processes a few part fields (like ref and value) when generating a PCB. I'll have to take a look at it.

I don't know if PCBNEW will allow you to update an existing PCB with the netlist having all your custom fields. You may just have to go with `generate_netlist()` and then import it into PCBNEW rather than using `generate_pcb()`.

tapegoji, 2024-08-27:
I faced the same issue. I think it would be a great feature to have. I am pretty familiar with kinet2pcb code. I can take a look as well and see if I can get it working that way. Let me know Dave. I don't want to do something if you are working on it already.

devbisme, 2024-08-27:
That would be a great help! If you do this, make your changes to the `development` branch of `kinet2pcb`.