devbisme/skidl #224: [SKiDL BUG] search_footprints results in `FileNotFoundError: [Errno 2] No such file or directory:`

blark, 2024-09-26:
**Describe the bug**
Search footprints, with a properly configured path doesn't work - resulting in an exception, seemingly due to improper parsing of the path in `footprint_search_paths` (running on macOS)

**To Reproduce**
Steps to reproduce the behavior:
1. set `footprint_search_paths['kicad8'] = '/Applications/KiCad/KiCad.app/Contents/SharedSupport/footprints/'`
2. search_footprints('cap')
3. FileNotFoundError: [Errno 2] No such file or directory: 'A'

**Expected behavior**
Return a list of footprints matching "cap" 

**Desktop (please complete the following information):**
 - OS macOS 14.5
 - Python version 3.12.3
 - SKiDL version [e.g. 22] 1.2.3

devbisme, 2024-09-26:
`footprint_search_paths['kicad8']` should be a list of paths where footprint files can be found, not a string for a single path:
```python
footprint_search_paths['kicad8'] = ['/Applications/KiCad/KiCad.app/Contents/SharedSupport/footprints/']
```

devbisme, 2024-09-27:
Hmm.

Did it work once you changed it to a list? If so, then there's a mistake in my initialization of the defaults.

devbisme, 2024-09-27:
I just checked and the `footprint_search_paths` entries are incorrectly initialized to strings instead of lists of strings. So I'll have to change that.