devbisme/skidl #227: [SKiDL BUG] asked for LM393, netlist and schematic shows LM2903

ziggy90127, 2024-10-26:
Kicad v5, macOS 10.14.6

```
#!/opt/local/bin/python3.12

import os

os.environ['KICAD_SYMBOL_DIR'] = '/Library/Application Support/kicad/library'

from skidl import *

u1 = Part('Comparator', 'LM393', ref='U1', footprint='Package_DIP:DIP-8_W7.62mm')

print(generate_netlist())
generate_schematic()
```

```
(export (version D)
  (design
    (source "sk2.py")
    (date "10/26/2024 11:05 AM")
    (tool "SKiDL (1.2.3)"))
  (components
    (comp (ref U1)
      (value LM2903)
      (footprint Package_DIP:DIP-8_W7.62mm)
      (fields
        (field (name F0) U)
        (field (name F1) LM2903))
      (libsource (lib Comparator) (part LM2903))
      (sheetpath (names /top/8858243136256944579) (tstamps /top/8858243136256944579))))
  (nets)
)
```
[image]

devbisme, 2024-10-29:
I looked into the Comparator.lib file for KiCad 5 and it appears that `LM393` is an alias for the `LM2903`, so the schematic is showing the main part identifier instead of the alias. Is this a mistake? If you drop an LM393 into the schematic, does it show `LM393` as the part name or `LM2903`?

```
#
# LM2903
#
DEF LM2903 U 0 5 Y Y 3 L N
F0 "U" 150 150 50 H V C CNN
F1 "LM2903" 250 -150 50 H V C CNN
F2 "" 0 0 50 H I C CNN
F3 "" 0 0 50 H I C CNN
ALIAS LM393
```

ziggy90127, 2024-10-29:
Thanks for looking into this. I created a new schematic in v5, and pressed A for add, and typed in LM393.  The description says it's an alias to LM2903.  The part it added is labelled LM393.  I was playing with chatgpt4o, and the netlist it generated included an LM393.  I hadn't considered that it may be an alias.  No matter, the circuit chatgpt4o generated was nonsense, so I'm not pursuing it.  Hopefully a future version of chatgpt will generate a .kicad_sch directly with no mistakes.

devbisme, 2024-10-29:
> I created a new schematic in v5, and pressed A for add, and typed in LM393. The description says it's an alias to LM2903. The part it added is labelled LM393.

OK, it sounds like SKiDL should generate the netlist using LM393 instead of using the LM2903 base part. I'll create an issue for this.

> Hopefully a future version of chatgpt will generate a .kicad_sch directly with no mistakes.

That may be a long time coming. Creating a schematic in a .kicad_sch file requires things like placing the part symbols and then routing the wires that connect the symbol pins (it's more complicated than making a netlist). I doubt there's enough examples to train chatgpt so that it can do this based only on statistics (chatgpt has a hard enough time doing arithmetic without using external tools). However, I would love to be wrong about this.

Having chatgpt write SKiDL (or something like it) and then generating the schematic using dedicated tools is probably going to be more feasible in the short term. chatgpt can already write Python and the entire SKiDL manual could be loaded into its context window to guide what it generates.

I'd be interested in hearing about any future experiments you do regarding circuit synthesis using AI tools. Just open a discussion thead.

closed 2024-11-15 by devbisme