devbisme/skidl #228: [SKiDL BUG] certain lines give skidl.schematics.route.RoutingFailure

ziggy90127, 2024-10-30:
I asked chatGPT4o to create a skidl.  I then did some editing to the code to get rid of some obvious blunders chatGPT4o made:
```
#!/opt/local/bin/python3.12

import os

# Set KiCad symbol directory
os.environ['KICAD_SYMBOL_DIR'] = '/Library/Application Support/kicad/library'

from skidl import *

set_default_tool(KICAD5)

# Define nets for power, ground, and other signals
gnd = Net("GND")
vcc = Net("VCC")
thermistor_signal = Net("Thermistor_Signal")  # Separate net for thermistor signal
comparator_out = Net("Comparator_Out")
timer_out = Net("555_Out")
zero_crossing = Net("Zero_Crossing")
ac_line = Net("AC_Line")

# Define components with specific references, values, and footprints
u1 = Part('Comparator', 'LM2903', ref='U1', footprint='Package_DIP:DIP-8_W7.62mm')
u2 = Part('Timer', 'NE555P', ref='U2', footprint='Package_DIP:DIP-8_W7.62mm')
u3 = Part('Isolator', 'H11AA1', ref='U3', footprint='Package_DIP:DIP-6_W7.62mm')
r1 = Part('Device', 'R', ref='R1', value='10k', footprint='Resistor_SMD:R_0805')
r2 = Part('Device', 'R', ref='R2', value='220', footprint='Resistor_SMD:R_0805')
r3 = Part('Device', 'R', ref='R3', value='10k', footprint='Resistor_SMD:R_0805')
c1 = Part('Device', 'C', ref='C1', value='100uF', footprint='Capacitor_THT:CP_Radial_D8.0mm_P3.5mm')
d1 = Part('Diode', '1N4148', ref='D1', value='1N4148', footprint='Diode_SMD:D_SOD-123')
rpot = Part('Device', 'R_POT', ref='Rpot', value='10k_Potentiometer', footprint='Potentiometer_Vertical:Potentiometer_3Pin')
ntc1 = Part('Device', 'Thermistor_NTC', ref='NTC1', value='10k', footprint='Resistor_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P7.62mm_Horizontal')
ssr1 = Part('Relay_SolidState', 'S102S01', ref='SSR1', footprint='Relay_THT:Solid_State_Relay')

# Connect components to nets as per the netlist

# GND connections
u1[4] += gnd
u2[1] += gnd
u3[2] += gnd
c1[2] += gnd
ntc1[2] += gnd
# rpot[2] += gnd

# # VCC connections
# u1[8] += vcc
u2[8] += vcc
# u3[1] += vcc
c1[1] += vcc
# rpot[1] += vcc  # Potentiometer connected to VCC
r1[1] += vcc  # Intermediate resistor R1 connected to VCC

# Thermistor signal net connections
r1[2] += thermistor_signal  # R1 connects thermistor_signal to VCC
ntc1[1] += thermistor_signal  # Thermistor connected to thermistor_signal
u1[3] += thermistor_signal  # Comparator input connected to thermistor_signal

# Comparator output to timer input
u1[1] += comparator_out
u2[2] += comparator_out

# Timer output to SSR control
u2[3] += timer_out
ssr1[4] += timer_out

# Zero-crossing detector connections
u3[4] += zero_crossing
u2[6] += zero_crossing

# AC Line connections for SSR
ssr1[1] += ac_line
ssr1[2] += ac_line

# Generate the netlist and print it
#print(generate_netlist())
generate_schematic()
```
The resulting schematic is:

ziggy90127, 2024-10-30:
```
pip-3.12 show skidl
Name: skidl
Version: 1.2.3
Summary: A Python package for textually describing electronic circuit schematics.
Home-page: https://github.com/devbisme/skidl
Author: Dave Vandenbout
Author-email: devb@xess.com
License: MIT
Location: /Users/dwiens/Library/Python/3.12/lib/python/site-packages
Requires: deprecation, future, graphviz, kinet2pcb, kinparse, sexpdata
Required-by: 
```

devbisme, 2024-11-03:
Try generating an SVG schematic. Going forward, I'm trying to emphasize that method rather than generating a KiCad schematic.

ziggy90127, 2024-11-03:
https://devbisme.github.io/skidl/ says to run "npm install https://github.com/nturley/netlistsvg", but I had to run "npm install https://github.com/nturley/netlistsvg -g" for it to work.  

Both the netlist and svg generation work...
[image]

closed 2024-11-03 by ziggy90127