devbisme/skidl #231: [SKiDL BUG] field values are not quoted correctly

cyberhuman, 2024-11-09:
**Describe the bug**

Field values are only quoted when they contain whitespaces or parentheses. This is a problem since KiCAD doesn't recognize unquoted values as valid, which results in empty fields when importing netlists generated by SKiDL.

[The description of the S-expression file format](https://dev-docs.kicad.org/en/file-formats/sexpr-intro/index.html) mentions all strings are quoted:
> All strings are quoted using the double quote character (") and are UTF-8 encoded.

This is also confirmed by [the source code of the netlist reader](https://gitlab.com/kicad/code/kicad/-/blob/4e04c05cd8f676149a6ab99e84196b1abe18c788/pcbnew/netlist_reader/kicad_netlist_reader.cpp#L426).

Additionally:
1. Double quotes and special characters are not escaped, resulting in broken netlist file.
2. Trying to work around the issue by quoting values doesn't work as the values are trimmed un-quoted by SKiDL.

**To Reproduce**
Steps to reproduce the behavior:

```py
from skidl import *

R = Part('Device', 'R', footprint='Resistor_SMD:R_0402_1005Metric')
R.fields['field1'] = 'unquoted_missing_from_kicad'
R.fields['field2'] = 'quoted present in kicad'
R.fields['field3'] = '"unexpectedly_unquoted_missing_from_kicad"'
R.fields['field4'] = ' "unexpectedly_trimmed_unquoted_missing_from_kicad" '
R.fields['field5'] = '"broken_netlist_file'

generate_netlist()
```

results in:
```lisp
...
      (fields
        (field (name field1) unquoted_missing_from_kicad)
        (field (name field2) "quoted present in kicad")
        (field (name field3) unexpectedly_unquoted_missing_from_kicad)
        (field (name field4) unexpectedly_trimmed_unquoted_missing_from_kicad)
        (field (name field5) "broken_netlist_file))
...
```

**Expected behavior**
I would expect values are always quoted and escaped. This is always safe to do at the expense of insignificant increase of file size.
I tested whether `\` works for escaping and it works for both `"` and for the newline character `\n`.

devbisme, 2024-11-10:
Thanks for the bug report. I pushed a possible fix to the `development` branch. Please try it and let me know if it solves the problem.
```
pip install git+https://github.com/devbisme/skidl@development
```

cyberhuman, 2024-11-11:
Hi @devbisme and thanks for the quick response!

Unfortunately while the fix works for field values, it now breaks field names:
```py
R.fields['field 6'] = 'broken_netlist_file'
```
results in
```lisp
        (field (name field 6) "broken_netlist_file"))
```

(it's real-world scenario: I have `JLCPCB Position Offset` and `JLCPCB Rotation Offset` for some parts)

devbisme, 2024-11-12:
I added double quotes on both names and values. Give it a try.

cyberhuman, 2024-11-17:
I have just tested it and it works! Thank you for the fix and keep up a great work!

commit 8afe4104 (2024-11-10): fix: Fix #231 for double-quoting strings and escaping internal quotes in KiCad netlists.
commit 68b85f8e (2024-11-12): fix: Fix #231 to add double quotes around field names.
closed 2024-11-20 by devbisme