devbisme/skidl #233: [SKiDL BUG] an error with function "generate_schematic"

d111e, 2024-12-04:
**Describe the bug**
i was able to use the function "generate_schematic" to generate .sch file in kicad 5, but now after i change all the key words "KICAD5" into "KICAD8" i got an error: "NameError: name 'active_logger' is not defined"
**To Reproduce**
Steps to reproduce the behavior:
1.simply by using the function "generate_schematic(tool=KICAD8)"

**Expected behavior**
it should be able to generate a .sch file

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS [type & version]: win 10
 - Python version: 3.8.20
 - SKiDL version [e.g. 22] 2.0.0

devbisme, 2024-12-04:
Thanks for reporting this error! I was able to re-create it as well. The error is triggered by this line:
```python
active_logger.warning("Schematic generation is not implemented for KiCad version 8.")
```
I will fix this error, but you can see from the error message that schematic generation isn't implemented for KiCad 8. So even without the error, you still can't generate a schematic.

d111e, 2024-12-05:
> Thanks for reporting this error! I was able to re-create it as well. The error is triggered by this line:
>
> ```python
> active_logger.warning("Schematic generation is not implemented for KiCad version [...]

ok, thanks for replying to my question, i guess i'll have to return to kicad 5, lol

devbisme, 2024-12-11:
After working for 1.5 years getting generating schematics for KiCad 5, I halted work for other versions because I couldn't see a viable design flow. For example, once you edited a generated schematic, how would you propagate those changes back into the SKiDL code?

Can you provide some details of how you're using SKiDL and the generated schematics?

closed 2024-12-12 by devbisme

d111e, 2024-12-19:
Sorry for keeping you waiting for so long, i was preparing for my exams and totally forgot to see my mailbox, again i express my apology to you. 

Now we are still trying to generate some schematics according to the simple image of the circuit, and we can't propagate any changes back to SKiDL code if something has been done by Kicad, what we do now is still at the beginning stage, and also the generated schematics seem very chaotic, the wires have high probability to be overlapped.

According to you, the error caused by the function seems to be related with Kicad 8 instead of SKiDL, then can you provide an example of generating schematic using SKiDL 2?

commit 1889bfb5 (2024-12-04): fix: Fix #233 by importing `active_logger` into schematic generation code for KiCad 6, 7, 8.