devbisme/skidl #244: [SKiDL BUG] lib_search_paths[KICAD].append() not finding local custom symbols

shanemmattner, 2025-02-02:
**Describe the bug**
I'm trying to use custom library components with SKIDL code and running into an error that SKIDL can't find the library with error:

```
ValueError: Unable to find part PT7C4563BQ1WEX in library UNKNOWN.
```

Code:
```python
lib_search_paths[KICAD].append('/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/')
```
I also tried specifying the file with the same results:
```python
lib_search_paths[KICAD].append('/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/skip_kicad_symbols.kicad_sym')
```

Terminal output:
```
venvShanes-MacBook-Pro:skidl shanemattner$ python3 example_control_board/control_board_SKIDL/main.py
ERROR: Can't open file: _skip.
 @ [/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py:19=>/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py:14]
WARNING: Could not load KiCad schematic library "_skip", falling back to backup library. @ [/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py:19=>/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py:14]
ERROR: Unable to find part PT7C4563BQ1WEX in library UNKNOWN. @ [/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py:19=>/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py:14]
Traceback (most recent call last):
  File "/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py", line 19, in <module>
    main()
    ~~~~^^
  File "/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/main.py", line 14, in main
    Project_Architecture()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/Users/shanemattner/Desktop/skidl/src/skidl/group.py", line 63, in sub_f
    results = f(*args, **kwargs)
  File "/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/Project_Architecture.py", line 32, in Project_Architecture
    ESP32S3(_p_3_3V, Project_Architecture_Coral_TPU_Coral_On, Project_Architecture_Coral_TPU_USB2_TPU_N, Project_Architecture_Coral_TPU_USB2_TPU_P, Project_Architecture_ESP32S3_BATT_CURRENT, Project_Architecture_ESP32S3_BATT_VOLTAGE, Project_Architecture_ESP32S3_LEG_PWR_1, Project_Architecture_ESP32S3_LEG_PWR_2, Project_Architecture_ESP32S3_RS485_DIR_1, Project_Architecture_ESP32S3_RS485_DIR_2, Project_Architecture_ESP32S3_UART1_RX, Project_Architecture_ESP32S3_UART1_TX, Project_Architecture_ESP32S3_UART2_RX, Project_Architecture_ESP32S3_UART2_TX, GND)
    ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shanemattner/Desktop/skidl/src/skidl/group.py", line 63, in sub_f
    results = f(*args, **kwargs)
  File "/Users/shanemattner/Desktop/skidl/example_control_board/control_board_SKIDL/ESP32S3.py", line 60, in ESP32S3
    IC1 = Part('_skip', 'PT7C4563BQ1WEX', value='PT7C4563BQ1WEX', footprint='_skip:SOIC127P600X175-8N', ref='IC1', fields={'Height': '1.75', 'Manufacturer_Name': 'Diodes Incorporated', 'Manufacturer_Part_Number': 'PT7C4563BQ1WEX', 'Mouser Part Number': '621-PT7C4563BQ1WEX', 'Mouser Price/Stock': 'https://www.mouser.co.uk/ProductDetail/Diodes-Incorporated/PT7C4563BQ1WEX?qs=P1JMDcb91o5VwflV1V69pA%3D%3D', 'Arrow Part Number': 'PT7C4563BQ1WEX', 'Arrow Price/Stock': 'https://www.arrow.com/en/products/pt7c4563bq1wex/diodes-incorporated?region=nac', 'Sheetname': 'ESP32S3', 'Sheetfile': 'esp32s3.kicad_sch'})
  File "/Users/shanemattner/Desktop/skidl/src/skidl/part.py", line 192, in __init__
    part = lib[name].copy(dest=TEMPLATE)
           ~~~^^^^^^
  File "/Users/shanemattner/Desktop/skidl/src/skidl/schlib.py", line 174, in __getitem__
    return list_or_scalar(self.get_parts_by_name(id))
                          ~~~~~~~~~~~~~~~~~~~~~~^^^^
  File "/Users/shanemattner/Desktop/skidl/src/skidl/schlib.py", line 265, in get_parts_by_name
    active_logger.raise_(ValueError, message)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shanemattner/Desktop/skidl/src/skidl/logger.py", line 131, in raise_
    raise exc_class(msg)
ValueError: Unable to find part PT7C4563BQ1WEX in library UNKNOWN.
venvShanes-MacBook-Pro:skidl shanemattner$ 
```

Here is the SKIDL python file were the error happens:
```python
# -*- coding: utf-8 -*-
from skidl import *

@subcircuit
def ESP32S3(_p_3_3V, Project_Architecture_Coral_TPU_Coral_On, Project_Architecture_Coral_TPU_USB2_TPU_N, Project_Architecture_Coral_TPU_USB2_TPU_P, Project_Architecture_ESP32S3_BATT_CURRENT, Project_Architecture_ESP32S3_BATT_VOLTAGE, Project_Architecture_ESP32S3_LEG_PWR_1, Project_Architecture_ESP32S3_LEG_PWR_2, Project_Architecture_ESP32S3_RS485_DIR_1, Project_Architecture_ESP32S3_RS485_DIR_2, Project_Architecture_ESP32S3_UART1_RX, Project_Architecture_ESP32S3_UART1_TX, Project_Architecture_ESP32S3_UART2_RX, Project_Architecture_ESP32S3_UART2_TX, GND):
    # Local nets
    Project_Architecture_ESP32S3_EN = Net('/Project Architecture/ESP32S3/EN')
    Project_Architecture_ESP32S3_EX_FLASH_HOLD = Net('/Project Architecture/ESP32S3/EX_FLASH_HOLD')
    Project_Architecture_ESP32S3_EX_FLSH = Net('/Project Architecture/ESP32S3/EX_FLSH')
    Project_Architecture_ESP32S3_EX_FLSH_WP = Net('/Project Architecture/ESP32S3/EX_FLSH_WP')
    Project_Architecture_ESP32S3_FET_TEMP = Net('/Project Architecture/ESP32S3/FET_TEMP')
    Project_Architecture_ESP32S3_HW_VER = Net('/Project Architecture/ESP32S3/HW_VER')
    Project_Architecture_ESP32S3_I2C_SCL = Net('/Project Architecture/ESP32S3/I2C_SCL')
    Project_Architecture_ESP32S3_I2C_SDA = Net('/Project Architecture/ESP32S3/I2C_SDA')
[... 155 lines cut ...]
    Net__U6_IO41_ += R27['2'], U6['34']
    Net__U6_IO42_ += R28['2'], U6['35']
    Net__U6_IO45_ += R29['2'], U6['26']
    Net__U6_IO46_ += R39['1'], U6['16']
    Net__U6_IO48_ += R18['2'], U6['25']
    Net__U6_TXD0_ += R24['2'], U6['37']
    Net__U7_SDO_slash_SA0_ += R35['1'], U7['1']
    return
```

**To Reproduce**
Steps to reproduce the behavior:
1. Define SKIDL part with custom local library part
2. Try to run SKIDL code as normal

**Expected behavior**
SKIDL should find the part/library and run SKIDL logic normally

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Macbook pro , OS 15.3
 - Python version: 3.13.1
 - SKiDL version 2.0.2

devbisme, 2025-02-02:
> `IC1 = Part('_skip', 'PT7C4563BQ1WEX', ...)`

It appears to be looking for a library named `_skip`. Is there a `_skip.kicad_sym` in the directory `/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/`?

> lib_search_paths[KICAD].append('/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/skip_kicad_symbols.kicad_sym')

This won't work because that's a path to a KiCad symbol file instead of to a directory. However, you could use the path to the symbol file directly in `Part()` like this:
```python
 IC1 = Part('/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/skip_kicad_symbols.kicad_sym', 'PT7C4563BQ1WEX'
```

shanemmattner, 2025-02-03:
> > `IC1 = Part('_skip', 'PT7C4563BQ1WEX', ...)`
> 
> It appears to be looking for a library named `_skip`. Is there a `_skip.kicad_sym` in the directory `/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/`?

The file path and name is:
```
/Users/shanemattner/Desktop/skip/electronics/PCB/pcb_libraries/kicad/skip_kicad_symbols.kicad_sym
```
I changed the library reference from `_skip` to `skip_kicad_symbols` then the SKIDL logic worked as expected. 

Thank you!

closed 2025-02-03 by shanemmattner