devbisme/skidl #246: [SKiDL BUG] `netlist_to_skidl` requires top sheet

shanemmattner, 2025-02-04:
In `development` branch the new logic for `netlist_to_skidl` will not properly parse parts or nets on the top sheet.  So you must have the top sheet just be documentation and other schematic sheets.  However, this logic works well for complex hierarchical sheets so we should leave it in for now.

Example of erroneous behavior:

Terminal command:
```
netlist_to_skidl -i simple.net -o simple_out
```

netlist file `simple.net`:
```
(export (version "E")
  (design
    (source "/Users/shanemattner/Desktop/skidl/simple/simple.kicad_sch")
    (date "2025-02-03T19:07:40-0800")
    (tool "Eeschema 8.0.8")
    (sheet (number "1") (name "/") (tstamps "/")
      (title_block
        (title)
        (company)
        (rev)
        (date)
        (source "simple.kicad_sch")
        (comment (number "1") (value ""))
        (comment (number "2") (value ""))
        (comment (number "3") (value ""))
        (comment (number "4") (value ""))
        (comment (number "5") (value ""))
        (comment (number "6") (value ""))
        (comment (number "7") (value ""))
        (comment (number "8") (value ""))
        (comment (number "9") (value ""))))
    (sheet (number "2") (name "/r2/") (tstamps "/f0106cc7-bcaf-4eae-80a7-294f7e8ff323/")
      (title_block
        (title)
        (company)
        (rev)
        (date)
        (source "r2.kicad_sch")
        (comment (number "1") (value ""))
        (comment (number "2") (value ""))
        (comment (number "3") (value ""))
        (comment (number "4") (value ""))
        (comment (number "5") (value ""))
        (comment (number "6") (value ""))
        (comment (number "7") (value ""))
        (comment (number "8") (value ""))
        (comment (number "9") (value "")))))
  (components
    (comp (ref "R1")
      (value "R")
      (description "Resistor")
      (fields
        (field (name "Footprint"))
        (field (name "Datasheet"))
        (field (name "Description") "Resistor"))
      (libsource (lib "Device") (part "R") (description "Resistor"))
      (property (name "Sheetname") (value "Root"))
      (property (name "Sheetfile") (value "simple.kicad_sch"))
      (property (name "ki_keywords") (value "R res resistor"))
      (property (name "ki_fp_filters") (value "R_*"))
      (sheetpath (names "/") (tstamps "/"))
      (tstamps "b877a166-ca19-4ecc-a7fb-fdc585ad6315"))
    (comp (ref "R3")
      (value "R")
      (description "Resistor")
      (fields
        (field (name "Footprint"))
        (field (name "Datasheet"))
        (field (name "Description") "Resistor"))
      (libsource (lib "Device") (part "R") (description "Resistor"))
      (property (name "Sheetname") (value "Root"))
      (property (name "Sheetfile") (value "simple.kicad_sch"))
      (property (name "ki_keywords") (value "R res resistor"))
      (property (name "ki_fp_filters") (value "R_*"))
      (sheetpath (names "/") (tstamps "/"))
      (tstamps "f52ce8ae-1b07-4abf-b779-2b61cb5c82e0"))
    (comp (ref "R2")
      (value "R")
      (description "Resistor")
      (fields
        (field (name "Footprint"))
        (field (name "Datasheet"))
        (field (name "Description") "Resistor"))
      (libsource (lib "Device") (part "R") (description "Resistor"))
      (property (name "Sheetname") (value "r2"))
      (property (name "Sheetfile") (value "r2.kicad_sch"))
      (property (name "ki_keywords") (value "R res resistor"))
      (property (name "ki_fp_filters") (value "R_*"))
      (sheetpath (names "/r2/") (tstamps "/f0106cc7-bcaf-4eae-80a7-294f7e8ff323/"))
      (tstamps "8d8ce9d4-7aee-497b-9789-d162dd2ef370")))
  (libparts
    (libpart (lib "Device") (part "R")
      (description "Resistor")
      (docs "~")
      (footprints
        (fp "R_*"))
      (fields
        (field (name "Reference") "R")
        (field (name "Value") "R")
        (field (name "Footprint"))
        (field (name "Datasheet") "~")
        (field (name "Description") "Resistor"))
      (pins
        (pin (num "1") (name "") (type "passive"))
        (pin (num "2") (name "") (type "passive")))))
  (libraries
    (library (logical "Device")
      (uri "/Applications/KiCad/KiCad.app/Contents/SharedSupport/symbols//Device.kicad_sym")))
  (nets
    (net (code "1") (name "+3V3")
      (node (ref "R1") (pin "1") (pintype "passive")))
    (net (code "2") (name "/r2/pwr_in")
      (node (ref "R1") (pin "2") (pintype "passive"))
      (node (ref "R2") (pin "1") (pintype "passive"))
      (node (ref "R3") (pin "1") (pintype "passive")))
    (net (code "3") (name "GND")
      (node (ref "R2") (pin "2") (pintype "passive"))
      (node (ref "R3") (pin "2") (pintype "passive")))))
```

`main.py` output:
```python
# -*- coding: utf-8 -*-
from skidl import *
from r2 import r2

def main():
    # Create global nets

    # Create subcircuits
    r2()

if __name__ == "__main__":
    main()
    generate_netlist()
```

`r2.py` output:
```
# -*- coding: utf-8 -*-
from skidl import *

@subcircuit
def r2():
    # Local nets
    r2_pwr_in = Net('/r2/pwr_in')
    GND = Net('GND')

    # Components
    R2 = Part('Device', 'R', value='R', ref='R2', fields={'Sheetname': 'r2', 'Sheetfile': 'r2.kicad_sch', 'ki_keywords': 'R res resistor', 'ki_fp_filters': 'R_*'})

    # Connections
    r2_pwr_in += R2['1']
    GND += R2['2']
    return

```

shanemmattner, 2025-02-04:
Correct, `kinparse` looks to be working properly.  I believe the `netlist_to_skidl` logic is failing due to the top Sheetname being `Root` vs the Sheetpath being `/`.

devbisme, 2025-02-24:
I reworked `netlist_to_skidl`. Now it will handle circuitry defined on the top level. I regenerated your example and compared the new SKiDL code to the original and they seem nearly equivalent. I imported the netlist generated by the new code into PCBNEW and that seemed OK. Still, I think you should take a look at what the new `netlist_to_skidl` is doing to make sure it's what you want.