devbisme/skidl #248: [SKiDL BUG] Using components that do not contain a defined datasheet property throws 'Key Error'

keithgh1, 2025-02-13:
**Describe the bug**
When custom components don't contain the datasheet property, then attempting to parse it throws an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Load a part that fits the description.

`zif = Part('220-3342-00-0602J.kicad_sym','220-3342-00-0602J')`

**Expected behavior**
I expected the part to load without error.

**Fix**

The problem exists in [(https://github.com/devbisme/skidl/blob/cad164fa6ea6bfdd872c6bd9c90a99e40c39b62e/src/skidl/tools/kicad8/lib.py#L42)]

Specifically

`    # Populate part fields from symbol properties. Properties will also be included below in drawing commands.
    props = {
        prop[1].lower(): prop
        for prop in part.part_defn
        if prop[0].value().lower() == "property"
    }
    part.ref_prefix = props["reference"][2]
    part.value = props["value"][2]
    part.fplist.append(props["footprint"][2])
    part.datasheet = props["datasheet"][2]
    part.draw_cmds[1].extend([props["reference"], props["value"]])`

Line 462: if props datasheet[2] doesn't exist, then the script errors out.

Replace line 462 with these two lines:

`    part.datasheet = props.get("datasheet", ["", "", ""])
    part.datasheet = part.datasheet[2] if len(part.datasheet) > 2 else ""`

I have no clue if this is the most robust or clean way of handling this, but I do know that this fixes my issue.

**Desktop (please complete the following information):**
 - Windows 11 Pro.
 - Jupyter Notebook 7.2.2
 - Python 3.12.7
 - SKiDL version 2.0.1

**Additional context**
This will likely need to be changed in all the different kicad version lib.py's. The troublesome part, downloaded from the normal part sites, is attached.

[220-3342-00-0602J.zip](https://github.com/user-attachments/files/18786187/220-3342-00-0602J.zip)

devbisme, 2025-02-15:
Thanks for the error report, location of the error, and the fix! I've made the corrections to the code in the `development` branch for KiCad 6, 7, and 8. The test suite passes, but my test data doesn't include any custom libraries with missing datasheet entries. So please try it out and see if it works for you. Also, let me know where I can get the custom library that causes this problem so I can add it to my test suite. Thanks!

keithgh1, 2025-02-15:
Hey! Thanks for the quick response! I attached the problematic library with the part to the issue, see zip!

I'll try this out in a bit.

keithgh1, 2025-02-15:
ok @devbisme I tested your change, works fine. Your change is more robust because it handles the other properties as well. Thanks for skidl -- having so much fun creating PCBs without schematics. Really speeding up the process!

Let me know if you need anything else.