devbisme/skidl #249: [SKiDL BUG] The pin function identifiers are not working.

tapegoji, 2025-02-14:
**Describe the bug**
When creating a part I want to add pins. When adding pins, I get the following Error:
`
"Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: type object 'Pin' has no attribute 'PASSIVE'

It seems like that Pin class no longer as pin_types attributes. Is this correct?
"`
**To Reproduce**
simply follow the tutorial and execute the following code:
`

my_lib = SchLib(name='my_lib')                      # Create an empty library object.
my_part = Part(name='R', tool=SKIDL, dest=TEMPLATE) # Create an empty part object template.
my_part.ref_prefix = 'R'                            # Set the part reference prefix.
my_part.description = 'resistor'                    # Set the part's description field.
my_part.keywords = 'res resistor'                   # Set the part's keywords.
my_part += Pin(num=1, func=Pin.PASSIVE)             # Add a pin to the part.
my_part += Pin(num=2, func=Pin.PASSIVE)             # Add another pin to the part.
my_lib += my_part                                   # Add the part to the library.

new_resistor = Part(my_lib, 'R')                    # Instantiate the part from the library.
my_lib.export('my_lib')     
`
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Creating the pin and assisting the func

**Desktop (please complete the following information):**
 - OS Ubuntu 22.04
 - Python version: 3.10
 - SKiDL version :2 development branch

**Additional context**
Add any other context about the problem here.

devbisme, 2025-02-14:
I changed some definitions in `pin.py` so now you have to access pin functions like this: `pin.pin_types.PASSIVE` or `skidl.pin.pin_types`. I suppose I could add back the old access method. Let me know if you need it.

tapegoji, 2025-02-14:
I think these new methods would do for me. It might be a good idea to add the old method back for to be compatible with the documentation and backward compatibility. Or changing the documentation.

closed 2025-02-14 by tapegoji