devbisme/skidl #251: [SKiDL Feature Request] add tag to fields automatically

cyberhuman, 2025-03-06:
**Is your feature request related to a problem? Please describe.**
When using tags and leaving out reference designator assignment to SKiDL, after importing the netlist into KiCAD's PCBnew, it often happens that it's not clear which component on the board corresponds to which component in the code, especially if there are many components of the same, with the same footprint, and with the same value.

**Describe the solution you'd like**
It would be useful to have each component's tag was visible in the list of fields. Then it would be easy to peek it quickly.

**Describe alternatives you've considered**
It's possible to add a field manually to each component but it feels cumbersome especially for frequently used components like resistors and capacitors.

**Additional context**
Add any other context or screenshots about the feature request here.

devbisme, 2025-03-07:
I think this should be possible by automatically placing the tag into the part fields when the netlist is generated. I'll try it and let you know.

devbisme, 2025-03-09:
Parts in a KiCad netlist now include fields "SKiDL tag" and "SKiDL Line" that contain the part's assigned tag and the file line where the part was instantiated.

Install the `development` branch to try this out.

cyberhuman, 2025-03-21:
@devbisme I just tested it and it works wonders, and very helpful! Thank you!
Just a small suggestion about the "SKiDL Line" field: IMO it should the file name should be relative and not absolute, at least by default. Maybe it's a matter of preference though.

devbisme, 2025-03-22:
Absolute file names:
* Are longer.
* Remain valid even if the netlist file is moved out of the project directory.
* Become invalid if the project directory is moved.
* Can be selected along with the line number and opened in an editor.

Relative file names:
* Are shorter.
* Become unusable if the netlist file is moved out of the project directory.
* Remain valid if the project directory is moved.
* It appears to be more difficult to select and open a file if it is a relative path.

What is the main problem you experience with absolute file names?

cyberhuman, 2025-03-23:
I'd like to add that absolute file names:
- Become unusable if the project is moved to a different directory. E.g. when moving it to a new directory structure or distributing the project to other users.
- Leak information about the system where the file was generated; in particular the user name and the directory structure.

These are the two main problems I have with absolute file names.

If this feature is there to remain, it would be nice if it was possible to turn it on or off (either opt-in or opt-out is ok).

cyberhuman, 2025-05-18:
Can the SKiDL Line functionality be made optional, please?
After using it for a while, I find it somewhat annoying that after adding one part in the beginning of the source file, all the remaining parts' properties change. It adds too much noise to the version control system.

devbisme, 2025-05-28:
I did the following:
- Added `track_src` attribute to `Circuit` class. Setting to `True` (the default) places tracking info into netlists about where parts are defined. Use `default_circuit.track_src = False` to turn off tracking of component source lines in netlists.
- Added `track_abs_path` attribute to `Circuit` class. Setting to `False` (the default) forces the use of relative file paths in tracking info. Setting to `True` allows absolute paths. Use `default_circuit.track_abs_path = False` to use relative file paths.
- Added the optional `track_src` parameter to `generate_netlist()`. If set `True` or `False`, it will override the value in the `Circuit.track_src` attribute. If it's not provided, the `Circuit.track_src` will be used.

These changes are in the `development` branch.

cyberhuman, 2025-06-07:
The parameters work, thank you!

For the relative paths, it seems they're generated relative to the location of SKiDL; not exactly what expected but hmm, is it possible to have paths relative to CWD or the file from where `generate_netlist()` was called?. 

I also figured there is another absolute path baked in the netlist by default, two times:
```sexp
(export
  (version "D")
  (design
    (source "/home/%%%%%%%%%%%%%%%%%%%%%%/skidl/./sch.py")
    (date "06/07/2025 07:59 PM")
    (tool "SKiDL (2.0.2)")
    (sheet
      (number 1)
      (name "/")
      (tstamps "/")
      (title_block
        (title)
        (company)
        (rev)
        (date)
        (source "/home/%%%%%%%%%%%%%%%%%%%%%%/skidl/./sch.py")
        (comment
          (number "1")
          (value ""))
...
```

devbisme, 2025-06-22:
The source file paths are now also controlled by the `track_abs_path` parameter.