devbisme/skidl #256: [SKiDL BUG]

roboPanda69, 2025-05-13:
**Describe the bug**
Not able to access footprints. When running the "search_footprints('QFN-48')" I am getting some errors on my system.

**To Reproduce**
Steps to reproduce the behavior:
1. Start `python`
2. Run the command - `from skidl import *`
3. Run the command - `search_footprints('QFN-48')`

**Expected behavior**
Should have returned something similar to this -
`Package_DFN_QFN: QFN-48-1EP_5x5mm_P0.35mm_EP3.7x3.7mm ("QFN, 48 Pin (https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf#page=38), generated with kicad-footprint-generator ipc_noLead_generator.py" - "QFN NoLead")
Package_DFN_QFN: QFN-48-1EP_5x5mm_P0.35mm_EP3.7x3.7mm_ThermalVias ("QFN, 48 Pin (https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf#page=38), generated with kicad-footprint-generator ipc_noLead_generator.py" - "QFN NoLead")
Package_DFN_QFN: QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm ("QFN, 48 Pin (https://static.dev.sifive.com/SiFive-FE310-G000-datasheet-v1p5.pdf#page=20), generated with kicad-footprint-generator ipc_noLead_generator.py" - "QFN NoLead")
Package_DFN_QFN: QFN-48-1EP_6x6mm_P0.4mm_EP4.2x4.2mm_ThermalVias ("QFN, 48 Pin (https://static.dev.sifive.com/SiFive-FE310-G000-datasheet-v1p5.pdf#page=20), generated with kicad-footprint-generator ipc_noLead_generator.py" - "QFN NoLead")`

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - Windows 11
 - Tried on multiple python version like Python 3.13 and Python 3.11
 - SKiDL version [2.0.1]

**Additional context**
I have added KICD8_SYMBOL_DIR and KICAD8_FOOTPRINT_DIR in the environment variable. But still it is not able to find it.

tapegoji, 2025-05-13:
Move to Linux and All you problem shall be resolved :-D

roboPanda69, 2025-05-13:
I tried on Linux too but I am still having the same error. Here is the screenshot - 

[image]

tapegoji, 2025-05-13:
@roboPanda69 It seems like that it's looking for your environment variables. try these and see if that works. I usually put them in .bashrc so I don't have to worry about them. 

export KICAD_SYMBOL_DIR=/usr/share/kicad/symbols/
export KICAD6_SYMBOL_DIR=/usr/share/kicad/symbols/
export KICAD7_SYMBOL_DIR=/usr/share/kicad/symbols/
export KICAD8_SYMBOL_DIR=/usr/share/kicad/symbols/
export KICAD_FOOTPRINT_DIR=/usr/share/kicad/footprints/
export KICAD6_FOOTPRINT_DIR=/usr/share/kicad/footprints/
export KICAD7_FOOTPRINT_DIR=/usr/share/kicad/footprints/
export KICAD8_FOOTPRINT_DIR=/usr/share/kicad/footprints/

roboPanda69, 2025-05-14:
Hi @tapegoji I had given set the environment variable in bashrc. But still no luck. 

[image]

I wanted to confirm a thing. Could you please tell what do you get when you run the commands (after importing skidl in python) - `print(config["lib_search_paths"]["kicad8"])` and `print(config["footprint_search_paths"]["kicad8"])`. I feel for my case the problem is coming here. 

When I run  `print(config["lib_search_paths"]["kicad8"])` I am getting `['.', '/usr/share/kicad/symbols/']`.

And When I run `print(config["footprint_search_paths"]["kicad8"])` I am getting `/home/prashun/.config/kicad`. And I checked there are no footprints in the directory `/home/prashun/.config/kicad`.

tapegoji, 2025-05-14:
I get /usr/local/share/kicad/symbols and footprints respectively. note that my kicad is in /usr/local.

roboPanda69, 2025-05-18:
HI I got the issue. So when before running the command like `search_footprints()` it is necessary to once open the `Footprint Editor` in KiCAD because after opening it for the first time then then the fp-lib-table would be created in the desired folder, which would later be referenced by our skidl code. Also there is another issue, in the current fp-lib-table library there is no component named `QFN-48` now.

closed 2025-05-18 by roboPanda69