devbisme/skidl #259: [SKiDL BUG] development branch - netlist can't be imported into KiCAD

cyberhuman, 2025-06-07:
**Describe the bug**
Netlist can't be imported into KiCAD. I figured it's because of KiCAD parser of S-expressions is not compliant and it expects whitespace after an atom and before the newline.

Arguably this should be reported to KiCAD developers, too, but it might be worth having a workaround in SKiDL for compatibility reasons.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate a netlist with SKiDL `development` branch.
2. In KiCAD's pcbnew, click File -> Import -> Netlist -> choose netlist file -> Load and Test Netlist
3. An error message is shown: "Cannot open netlist file '<filename>'."

**Expected behavior**
Netlist file must be accepted by pcbnew.

**Desktop (please complete the following information):**
SKiDL version: development branch, 714497a7e7056ef60a9d08a1907d99044cf8cde7

**Additional context**
I found two workarounds for the issue:
1. Add a trailing whitespace to every line of the netlist.
2. Remove all newlines.

devbisme, 2025-06-07:
OK, I'll make this a priority.

devbisme, 2025-06-07:
It was a whitespace problem as you indicated. The problem can be corrected by adding a single space after the `(export` on the first line. I'll let you know when it's fixed.

cyberhuman, 2025-06-08:
Yes it works now! Thank you for the quick fix.

closed 2025-06-08 by cyberhuman