devbisme/skidl #264: Behavioral voltage source definition not working

mbacvanski, 2025-07-23:
**Describe the bug**
Behavioral voltage sources throw an exception when you try to define them

**To Reproduce**
```python
from skidl import *
from skidl.pyspice import *

reset()

# Make two test nets
n1, n2 = Net('N1'), Net('N2')

# Add a behavioral source B1 between N1 and N2, trying to set its voltage to a sine wave
b = B(v="sin(2*pi*1k*time)", footprint=None)
b[1, 2] += n1, n2

print(generate_netlist(tool=SPICE))
```

Raises an error 
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[15], line 15
     12 b[1, 2] += n1, n2
     14 # Print out the SPICE netlist
---> 15 print(generate_netlist(tool=SPICE))

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/skidl/circuit.py:495, in Circuit.generate_netlist(self, **kwargs)
    492 file_ = kwargs.pop("file_", None)
    493 do_backup = kwargs.pop("do_backup", True)
--> 495 netlist = tool_modules[tool].gen_netlist(self, **kwargs)
    497 active_logger.report_summary("generating netlist")
    499 if not self.no_files:

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/skidl/tools/spice/spice.py:360, in gen_netlist(self, **kwargs)
    358         active_logger.error("Part has no SPICE model: {}".format(part))
    359     else:
--> 360         add_func(part, circuit)
    362 return circuit

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/skidl/tools/spice/spice.py:467, in add_part_to_circuit(part, circuit)
    464     pass
    466 # Add the part to the PySpice circuit.
--> 467 getattr(circuit, part.pyspice["name"])(*args, **kwargs)

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/PySpice/Spice/__init__.py:56, in _make_function.<locals>.function(self, *args, **kwargs)
     55 def function(self, *args, **kwargs):
---> 56     return element_class(self, *args, **kwargs)

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/PySpice/Spice/Netlist.py:691, in FixedPinElement.__init__(self, netlist, name, *args, **kwargs)
    688             raise NameError("Node '{}' is missing for element {}".format(pin_definition.name, self.name))
    689         pin_definition_nodes.append((pin_definition, node))
--> 691 super().__init__(netlist, name, *args, **kwargs)
    693 self._pins = [Pin(self, pin_definition, netlist.get_node(node, True))
    694               for pin_definition, node in pin_definition_nodes]

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/PySpice/Spice/Netlist.py:526, in Element.__init__(self, netlist, name, *args, **kwargs)
    524         pass # cf. NonLinearVoltageSource
    525     else:
--> 526         raise ValueError('Unknown argument {}={}'.format(key, value))
    528 self._pins = ()
    529 netlist._add_element(self)

ValueError: Unknown argument v_expression=sin(2*pi*1k*time)
```

If I use the keyword parameter `value` or `v_expression` or anything else instead of `v`, the generated netlist still never contains the expression for the voltage source:
```
.title 
B1 N1 N2
```

**Desktop (please complete the following information):**
 - MacOS Sequoia 15.5
 - Python 3.10
 - SKiDL version 2.0.1

Maybe I'm using the wrong syntax, but I couldn't find any examples or documentation on behavioral voltage sources.

mbacvanski, 2025-07-24:
Hmm I installed the development branch with that command but am now getting this error

```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Cell In[1], line 15
     12 b[1, 2] += n1, n2
     14 # Print out the SPICE netlist
---> 15 print(generate_netlist(tool=SPICE))

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/skidl/circuit.py:681, in Circuit.generate_netlist(self, **kwargs)
    678 kwargs["track_abs_path"] = kwargs.get("track_abs_path", self.track_abs_path)
    679 kwargs["track_src"] = kwargs.get("track_src", self.track_src)
--> 681 netlist = tool_modules[tool].gen_netlist(self, **kwargs)
    683 active_logger.report_summary("generating netlist")
    685 if not self.no_files:

File ~/mambaforge/envs/AnalogAI/lib/python3.10/site-packages/skidl/tools/spice/spice.py:309, in gen_netlist(self, **kwargs)
    307 # Create an empty PySpice circuit.
    308 title = kwargs.pop("title", "")  # Get title and remove it from kwargs.
--> 309 circuit = PySpiceCircuit(title)
    311 # Default SPICE libraries will be read-in down below if needed.
    312 default_libs = []

NameError: name 'PySpiceCircuit' is not defined
```

devbisme, 2025-07-25:
The immediate fix for this problem is to upgrade your Python to version 3.11 or greater.

SKiDL originally used PySpice to interface to a SPICE executable like ngspice. But PySpice hasn't been updated since 2020 and doesn't support newer versions of ngspice. Somebody forked PySpice and created InSpice which they updated to run modern versions of ngspice, but InSpice doesn't run on anything older than Python 3.11. The SKiDL installer won't install InSpice for older Python interpreters, so the PySpiceCircuit class will be undefined as happened to you. SKiDL fails silently in this case which should be corrected and possibly fallback to using PySpice, but that's not what currently happens.

tapegoji, 2025-07-25:
Behavioral voltage sources have limited support in InSpice and even PySpice. If this is something related to InSpice (in case you upgrade your python) then I'd be happy to help on the InSpice side.

mbacvanski, 2025-07-26:
Gotcha, the development branch works great after upgrading to Python 3.12. Thank you for the help!

closed 2025-11-11 by devbisme